Integrated-Project-2-2021-UVic-UCC/power-glove #21: Mid-term Presentation

ibraamar, 2021-03-24:
Here you have the mid-term presentation. If you have any doubt, contact us !

[Mid-Term.pdf](https://github.com/Integrated-Project-2-UVic-UCC/power-glove/files/6198856/Mid-Term.pdf)

nincodinachs2000, 2021-03-26:
You did it very well at the mid-term presentation, but I have something to say about it. The objectives were very clear and rationals, also the project is very innovative. But will the prototype mechanic be able to support and charge with the microcontroler, the components to feed it self and the drivers if are they needed? Also are you going to distribute the weight only to the wrist, or are you going to distribute it?

marcsala27, 2021-03-26:
Hey, I really liked your presentation and I feel that it's going to be a very good and innovative project with grate results. Although the presentation was so clear, I have a question, how are you going to implement the translated words or letters? Are you using an application? Thank you!

SergiMuac, 2021-03-27:
Hi everyone!!

First of all congratulations on your job!!
Your presentation was quite clear but you ran out of time so for the next one bear in mind to be briefer.
We didn't understand your objectives. We don't know if you don't know which are your objectives or you just don't know how to write them. The point is that an objective must be concrete and measurable so at the end you can see if you have completed it.
On the other hand, we don't believe that the mechanical part will be comfortable when the glove is worn, but we will wait to test the final prototype to decide on it.

To sum up, we think your project is very complex, but it seems to be well on track!

Eye-Sphere Team.

Guillem-Maza, 2021-03-28:
After attending your presentation our group agree that you have preperated it, but maybe individually, because as all of us know the time was not so adjusted.  
We valorate positively the way you work because it’s maybe more difficult to communicate without the possibility of be face to face of all the members.  
All the things you present look well worked, so we believe that you are going to get the goals you mark in the first of the project.  
We didn’t understood the function of the mechanic part that you have already made in the slide 12 of the presentation. 

Keep working and good luck! 
Covid Device team.

josepmaria-carbonell, 2021-04-06:
**Good job guys!**

The presentation was difficult to manage because of the online coordination between team members.  I think your message was received by the attendance. However, you must be proactive when explaining.
Please answer the formulated questions raised above by the other teams.

@Integrated-Project-2-UVic-UCC/power-glove

omartinezc, 2021-04-06:
> You did it very well at the mid-term presentation, but I have something to say about it. The objectives were very clear and rationals, also the project is very innovative. But will the prototype [...]

Nice question. The mechanic device that we want to build will be connected via Port Serial to the PC, so, batteries are not needed. But, in case that we create a wireless device, this must be able to charge enought energy througt rechargeable batteries (we don't rule it out).

About the second question, we will distribute all the weight of the electronics on the wrist, as the components weight so few as a regular wrist watch.

closed 2021-04-07 by omartinezc

gervidtab, 2021-04-14:
> Hey, we really liked your presentation and we feel that it's going to be a very good and innovative project with great results. Although the presentation was so clear, we have a question, how are you going to implement the translated words or letters? Are you using a mobile application?
> Thank you!

Hi Marc, thanks for the feedback!
The translated words will be translated through the interpretation of the hand position (six degrees of freedom) and fingers (five degrees of freedom)  and it will be reproduced on the computer screen.

About the question of the mobile app, by the moment we haven't considered it, but it's a very innovative idea for the future of our project!

gervidtab, 2021-04-14:
> Hi everyone!!
>
> First of all congratulations on your job!!
> Your presentation was quite clear but you ran out of time so for the next one bear in mind to be briefer.
> We didn't understand your [...]

Hello!
First of all, thank you for your advices, at the next presentation we will procure not to pass the time.
Regarding about the objectives, we haven't seen it like this since we have read your feedback, so thanks for it and we will take the relevant measures.

About the mechanical part, we saw what you say when we had printed the first design, and we also mentioned it in the mid-term presentation, so we have to tell you that we are making improvings in the mechanical design to be more comfortable and ergonomic on the wrist.

gervidtab, 2021-04-14:
> After attending your presentation our group agree that you have preperated it, but maybe individually, because as all of us know the time was not so adjusted.
> We valorate positively the way you work [...]

Hello! Thank you for your feedback, and yes we confirm that the communication is more dificult without the possibility of be face to face.
About the slide 12, we wanted to explain that we did a first design, and following the methodology of PLAN - DO - CHECK - ACT, when we saw that the first design was too weighted and heighted, we designed another piece more ergonomic and light.