Integrated-Project-2-2021-UVic-UCC/power-glove #23: Final shopping list

omartinezc, 2021-04-07:
Hello team, here we will post all the material recived and the remaining to obtain. Also to add new materials if needed.

gervidtab, 2021-04-12:
[image]
Arduino nano.

gervidtab, 2021-04-12:
[image]

ARDUINO NANO with an implanted PCB below it and five resistances of 10kΩ

gervidtab, 2021-04-12:
[image]
MPU9250.

gervidtab, 2021-04-12:
[image]
Resistances of 10kΩ.

gervidtab, 2021-04-12:
[image]
GLOVE

gervidtab, 2021-04-12:
[image]
FLEXSENSORS.

gervidtab, 2021-04-12:
[image]
three flexsensors of 2.18 inches and two of 3.75 inches. Provider: Spectra Symbol.

gervidtab, 2021-04-12:
[image]
PLA 3D filament. Provider: Smart Materials 3D.

gervidtab, 2021-04-12:
4 screws metric 2
2 pairs of power strips with pins male-male
2 pairs of power strips with pins male-female

closed 2021-06-05 by ibraamar